GloDroid/glodroid_manifest #197: GloDroid v0.7.4 on PinePhone v1.2: phone claims to be encrypted after installation, even when no password or PIN is configured yet

volkertb, 2022-04-23:
See my complete list of test findings (with some extra info) in this discussion: https://github.com/GloDroid/glodroid_manifest/discussions/195

volkertb, 2022-05-17:
Tested this with v0.7.5. It has the same issue.

herbsmn, 2022-11-03:
Tested with v0.8.1 and the issue is still happening.

rsglobal, 2022-11-04:
GloDroid has FBE (File-Based Encryption) enabled by default for all boards. Probably the 'encrypted' label related to that?

What is the expected behavior?

herbsmn, 2022-11-04:
You're correct. It looks like @volkertb and I were expecting it to be like full-disk encryption, which was disabled in Android 13. https://source.android.com/docs/security/features/encryption/full-disk

Seems like this issue can be closed.

closed 2022-11-20 by rsglobal