GloDroid/glodroid_manifest #198: GloDroid v0.7.4 on PinePhone v1.2: occasional reboots while streaming a podcast with AntennaPod

volkertb, 2022-04-23:
Reporting another finding: the phone occasionally reboots when I stream a podcast in the [AntennaPod](https://f-droid.org/en/packages/de.danoeh.antennapod/) app. (Note that his happened while the podcast was streaming. On this phone with this OS installed, I haven't yet tried playing back a podcast after first downloading it.)

It's happened twice already, a half hour into the episode.

Detailed information to help reproduce this:

- OS version: GloDroid v0.7.4
- Device: PinePhone v1.2 with 2GB RAM ("UBports Community Edition")
- AntennaPod version: 2.5.1 (installed with F-Droid)
- Podcast name: The WAN Show
- Episode name: An Update On My fraud (15 April 2022)

volkertb, 2022-04-24:
This just happened to me again. During this occurrence, I noticed that the screen actually freezes for a bit (a fraction of a second) after the audio cuts off, after which the phone reboots. Perhaps some kind of watchdog timer triggers a reboot after a crash?

herbsmn, 2022-11-04:
I have downloaded a few podcasts via AntennaPod on my PP with GloDroid v0.8.1 and playback after download didn't result in any issues at all.

I'll try streaming a few podcasts instead of downloading them, now that I see you had issues doing so, and will report back.

herbsmn, 2022-11-04:
Last night I streamed an entire 2 hour podcast with AntennaPod and experienced zero reboots or audio cut offs. 

I think this ticket can be closed since it doesn't seem to be reproducible in the most recent build.

herbsmn, 2022-11-17:
I have streamed about 10 more podcasts without any reboots or audio cut offs. I recommend closing this issue.

rsglobal, 2022-11-20:
@herbsmn ,

Thank you for testing.

closed 2022-11-20 by rsglobal