GloDroid/glodroid_manifest #201: Failed to run PinephonePro image on rk3399-leez-p710

andyshrk, 2022-04-30:
Much thanks to GloDroid，it‘s a nice project。
I want to run GloDroid on rk3399（rk3399-leez-p710， which also in mainline kernel） based board for a long time，I  have tried  run with opi4 image last year，but it failed due to some Serror。
When I see PPP support lunched，I want to try again。 
When flash deploy-sd.img to a sdcard ，and board on the board，I can see the boot log from the uart debug console，and can confirm the board run into recover mode，but it seems that recovery founction run failed：
```
# [  114.373091] read descriptors
[  114.373540] read strings
[  114.376585] init: processing action (sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1) from (
/system/etc/init/hw/init.rc:175)
[  114.395812] init: Command 'symlink /config/usb_gadget/g1/functions/ffs.adb /config/usb_gadget/g1/configs/b.1
/f1' action=sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:178) t
ook 5ms and failed: symlink() failed: File exists
[  114.398731] init: Command 'write /config/usb_gadget/g1/UDC ${sys.usb.controller}' action=sys.usb.config=adb 
&& sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:179) took 0ms and failed: property 's
ys.usb.controller' doesn't exist while expanding '${sys.usb.controller}'
rebo[  115.397524] read descriptors
[  115.397888] read strings
[  115.400905] init: processing action (sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1) from (
/system/etc/init/hw/init.rc:175)
[  115.420112] init: Command 'symlink /config/usb_gadget/g1/functions/ffs.adb /config/usb_gadget/g1/configs/b.1
/f1' action=sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:178) t
ook 5ms and failed: symlink() failed: File exists
[  115.422970] init: Command 'write /config/usb_gadget/g1/UDC ${sys.usb.controller}' action=sys.usb.config=adb 
&& sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:179) took 0ms and failed: property 's
ys.usb.controller' doesn't exist while expanding '${sys.usb.controller}'
ot fa[  116.425077] read descriptors
[  116.425516] read strings
[  116.428534] init: processing action (sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1) from (
/system/etc/init/hw/init.rc:175)
[  116.447790] init: Command 'symlink /config/usb_gadget/g1/functions/ffs.adb /config/usb_gadget/g1/configs/b.1
/f1' action=sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:178) t
ook 5ms and failed: symlink() failed: File exists
[  116.450711] init: Command 'write /config/usb_gadget/g1/UDC ${sys.usb.controller}' action=sys.usb.config=adb 
&& sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:179) took 0ms and failed: property 's
ys.usb.controller' doesn't exist while expanding '${sys.usb.controller}'
stboot[  117.449518] read descriptors
[  117.449881] read strings
[  117.452911] init: processing action (sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1) from (
/system/etc/init/hw/init.rc:175)

[  117.473131] init: Command 'symlink /config/usb_gadget/g1/functions/ffs.adb /config/usb_gadget/g1/configs/b.1
/f1' action=sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:178) t
ook 5ms and failed: symlink() failed: File exists
[  117.476125] init: Command 'write /config/usb_gadget/g1/UDC ${sys.usb.controller}' action=sys.usb.config=adb 
&& sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:179) took 0ms and failed: property 's
ys.usb.controller' doesn't exist while expanding '${sys.usb.controller}'
[  117.514619] init: Received sys.powerctl='reboot,fastboot' from pid: 317 (reboot)
[  117.515555] init: sys.powerctl: do_shutdown: 0 IsShuttingDown: 0
[  117.517249] init: Got shutdown_command 'reboot,fastboot' Calling HandlePowerctlMessage()
[  117.523895] init: Failed to set bootloader message: failed to open /dev/block/by-name/misc: No such file or 
directory
[  118.473088] read descriptors
[  118.473526] read strings
[  118.476594] init: processing action (sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1) from (
/system/etc/init/hw/init.rc:175)
[  118.495860] init: Command 'symlink /config/usb_gadget/g1/functions/ffs.adb /config/usb_gadget/g1/configs/b.1
/f1' action=sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:178) t
ook 5ms and failed: symlink() failed: File exists
[  118.498744] init: Command 'write /config/usb_gadget/g1/UDC ${sys.usb.controller}' action=sys.usb.config=adb 
&& sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:179) took 0ms and failed: property 's
ys.usb.controller' doesn't exist while expanding '${sys.usb.controller}'
[  119.497620] read descriptors
[  119.497983] read strings
[  119.501324] init: processing action (sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1) from (
/system/etc/init/hw/init.rc:175)
[  119.520497] init: Command 'symlink /config/usb_gadget/g1/functions/ffs.adb /config/usb_gadget/g1/configs/b.1
/f1' action=sys.usb.config=adb && sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:178) t
ook 5ms and failed: symlink() failed: File exists
[  119.523453] init: Command 'write /config/usb_gadget/g1/UDC ${sys.usb.controller}' action=sys.usb.config=adb 
&& sys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:179) took 0ms and failed: property 's
ys.usb.controller' doesn't exist while expanding '${sys.usb.controller}'
```

It run with the sys.usb .... loop and loop, never run the real recovery function and reboot.

The full log is here【0】
I don't know if this is something with  the dts file, I will try run with leez-p710 dtb later.
Any help would be appreciate。

I even fore reboot the board from recovery console, then stop it at u-boot console, then run `fastboot 0` fore the board enter
fastboot mode,  and run the following flash command 
```
./fastboot flash boot   boot.img
./fastboot flash super  super.img

./fastboot format:ext4 userdata
./fastboot format:ext4 metadata

./fastboot reboot
```
But failed at  format userdata metadata command,  I guess it beacuse the recovery not run success.

【0】https://pastebin.com/WVETEDJ2

rsglobal, 2022-04-30:
Hi,

> Failed to set bootloader message: failed to open /dev/block/by-name/misc: No such file or 
directory

This message can appear in case you have not provided correct MMC sysfs paths.
i.e. https://github.com/GloDroid/glodroid_device/blob/36e6b427e9b7de9bd9fd7588723a1733045453a8/opi3/opi3.mk#L30

Paths can be obtained from .dts file or by looking through recovery mode console in /dev/block/platform directory if possible.

andyshrk, 2022-04-30:
> Much thanks to GloDroid，it‘s a nice project。 I want to run GloDroid on rk3399（rk3399-leez-p710， which also in mainline kernel） based board for a long time，I have tried run with opi4 image last [...]

After run gd-v0.7.4-rpi4-userdebug.tar.gz on RPI4, I realized the recovery depends  usb adb to flash last super.img and do userdata format， so I need to figure out why my P710 sys.usb failed here.

> ```
>
> It run with the sys.usb .... loop and loop, never run the real recovery function and reboot.
>
> The full log is here【0】 I don't know if this is something with the dts file, I will try run with [...]

rsglobal, 2022-04-30:
Hm... I can't see any mention of fe320000.mmc initialization in your dmesg log. 

Can you try flashing emmc?

andyshrk, 2022-05-01:
> Hm... I can't see any mention of fe320000.mmc initialization in your dmesg log.
> 
> Can you try flashing emmc?

when try with emmc:
1. flash deploy-sd-for-emmc.img to sdcard
2. insert sdcard to board, and power on
3. run flash-emmc.sh script...

```
andy@B150:/srv/WorkSpace/GloDroid/out/target/product/pinephonepro$ ./flash-emmc.sh 
+ ./fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.009s]
Writing 'gpt'                                      OKAY [  0.026s]
Finished. Total time: 0.122s
+ ./fastboot flash bootloader bootloader-emmc.img
Sending 'bootloader' (16352 KB)                    OKAY [  0.909s]
Writing 'bootloader'                               OKAY [  1.136s]
Finished. Total time: 2.090s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.016s]
Writing 'uboot-env'                                OKAY [  0.031s]
Finished. Total time: 0.082s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.001s]
Finished. Total time: 0.252s
+ set +x
Power-down, remove recovery SD-CARD, and power-up the board.
Press enter to continue.

+ ./fastboot oem format
OKAY [  0.076s]
Finished. Total time: 0.076s
+ ./fastboot flash bootloader bootloader-emmc.img
Sending 'bootloader' (16352 KB)                    OKAY [  0.892s]
Writing 'bootloader'                               OKAY [  1.134s]
Finished. Total time: 2.088s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.016s]
Writing 'uboot-env'                                OKAY [  0.031s]
Finished. Total time: 0.113s
+ ./fastboot flash recovery_boot boot.img
Sending 'recovery_boot' (65536 KB)                 OKAY [  3.549s]
Writing 'recovery_boot'                            OKAY [  4.539s]
Finished. Total time: 8.159s
+ ./fastboot flash dtbo_a boot_dtbo.img
Sending 'dtbo_a' (72 KB)                           OKAY [  0.007s]
Writing 'dtbo_a'                                   OKAY [  0.037s]
Finished. Total time: 0.128s
+ ./fastboot erase misc
Erasing 'misc'                                     OKAY [  0.027s]
Finished. Total time: 0.130s
+ ./fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.035s]
< waiting for any device >
```

There is still no lucky, the system stop in recovery whith the flowing log:
```
[    6.152068] read strings
[    6.153302] init: processing action (sys.usb.config=fastboot && sys.usb.configfs=1 && sys.usb.ffs.ready=1) from (/sys
tem/etc/init/hw/init.rc:182)
[    6.153856] fastbootd: [libfs_mgr]unknown volume for path [/cache]
[    6.158193] init: Command 'write /config/usb_gadget/g1/UDC ${sys.usb.controller}' action=sys.usb.config=fastboot && s
ys.usb.configfs=1 && sys.usb.ffs.ready=1 (/system/etc/init/hw/init.rc:186) took 0ms and failed: property 'sys.usb.contro
ller' doesn't exist while expanding '${sys.usb.controller}'

# ls /sys/class/u                                                              
udc/       usb_role/
# ls /sys/class/udc/                                                           
# [   47.866571] random: crng init done
[   47.866879] random: 7 urandom warning(s) missed due to ratelimiting
```

I can see the fastbootd ui on hdmi screen, but the fastboot is not start.
I still can't figure out why   ${sys.usb.controller} is not set, and /sys/class/udc/ is empty

Here is the full log:
[1]https://pastebin.com/UJNLGeDt

andyshrk, 2022-05-01:
The emmc(sdhci: mmc@fe330000 ) and  sdmmc(sdmmc: mmc@fe320000 ) works fine on my board, I can see they find the correct media volume an partation from kernel log

rsglobal, 2022-05-01:
You can try playing with device tree `dr_mode="host/otg/peripheral"` strings.

Also ensure otg/usb module is compiled as built_in, not a kernel module.

andyshrk, 2022-05-01:
> You can try playing with device tree `dr_mode="host/otg/peripheral"` strings.
> 
> Also ensure otg/usb module is compiled as built_in, not a kernel module.

Yes, my colleague also suggests setting dr_mode to peripheral, I will try that later.

I also found that the mainline dts of p710 missing the fusb302 part, maybe that is also the case.

andyshrk, 2022-05-01:
Setting dr_mode to peripheral make adb/fastboot work okay, So the whole flash work can go on.

Now the board run into android boot process, but it seems that it can't run into android launcher, I can only see the andoid
boot animation loop again and again。

```
:s0 tclass=file permissive=1
[ 2120.413924] init: starting service 'exec 778 (/system/bin/flags_health_check UPDATABLE_CRASHING)'...
[ 2120.416581] type=1400 audit(1358501444.280:38720): avc: denied { open } for comm="flags_health_ch" path="/dev/__prope
rties__/u:object_r:lpdumpd_prop:s0" dev="tmpfs" ino=203 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:lpdumpd_p
rop:s0 tclass=file permissive=1
[ 2120.418952] type=1400 audit(1358501444.280:38721): avc: denied { getattr } for comm="flags_health_ch" path="/dev/__pr
operties__/u:object_r:lpdumpd_prop:s0" dev="tmpfs" ino=203 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:lpdump
d_prop:s0 tclass=file permissive=1
[ 2120.421418] type=1400 audit(1358501444.280:38722): avc: denied { map } for comm="flags_health_ch" path="/dev/__proper
ties__/u:object_r:lpdumpd_prop:s0" dev="tmpfs" ino=203 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:lpdumpd_pr
op:s0 tclass=file permissive=1
[ 2120.423559] type=1400 audit(1358501444.280:38723): avc: denied { open } for comm="flags_health_ch" path="/dev/__prope
rties__/u:object_r:media_config_prop:s0" dev="tmpfs" ino=204 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:medi
a_config_prop:s0 tclass=file permissive=1
[ 2120.424265] init: SVC_EXEC service 'exec 778 (/system/bin/flags_health_check UPDATABLE_CRASHING)' pid 8828 (uid 1000 
gid 1000+0 context default) started; waiting...
[ 2120.425763] type=1400 audit(1358501444.280:38724): avc: denied { getattr } for comm="flags_health_ch" path="/dev/__pr
operties__/u:object_r:media_config_prop:s0" dev="tmpfs" ino=204 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:m
edia_config_prop:s0 tclass=file permissive=1
[ 2120.429214] type=1400 audit(1358501444.280:38725): avc: denied { map } for comm="flags_health_ch" path="/dev/__proper
ties__/u:object_r:media_config_prop:s0" dev="tmpfs" ino=204 scontext=u:r:flags_health_check:s0 tcontext=u:object_r:media
_config_prop:s0 tclass=file permissive=1
[ 2120.457129] init: Service 'exec 778 (/system/bin/flags_health_check UPDATABLE_CRASHING)' (pid 8828) exited with statu
s 0 waiting took 0.038000 seconds
[ 2120.458377] init: Sending signal 9 to service 'exec 778 (/system/bin/flags_health_check UPDATABLE_CRASHING)' (pid 882
8) process group...
[ 2120.459741] libprocessgroup: Successfully killed process cgroup uid 1000 pid 8828 in 0ms
[ 2120.640132] init: Control message: Could not find 'aidl/android.hardware.vibrator.IVibrator/default' for ctl.interfac
e_start from pid: 314 (/system/bin/servicemanager)
[ 2121.642427] init: Control message: Could not find 'aidl/android.hardware.vibrator.IVibrator/default' for ctl.interfac
e_start from pid: 314 (/system/bin/servicemanager)
[ 2122.643724] init: Control message: Could not find 'aidl/android.hardware.vibrator.IVibrator/default' for ctl.interfac
e_start from pid: 314 (/system/bin/servicemanager)
```

The full log is here[2]

I am not sure if my dts is too simple that lack somewhat device that makes some critical android service run fail.
[2]https://pastebin.com/8qUp3ecZ

rsglobal, 2022-05-01:
Please use `adb logcat` to get more logs.

rsglobal, 2022-05-01:
Feel free to attach logcat output if you need any help with analysis.

andyshrk, 2022-05-03:
> Feel free to attach logcat output if you need any help with analysis.
Thank you  very much for you help.
I first want to past logcat but i found it very big.

After enable a `CONFIG_TEST_POWER` and a fake vibrator dt node at p710 dts, the board can now boot into launcher.
P710 is a SBC, so there is no VIBRATOR, no battery, no sensor, no modem,  maybe I should remove these service in device.mk ?

rsglobal, 2022-05-03:
> After enable a CONFIG_TEST_POWER and a fake vibrator dt node at p710 dts, the board can now boot into launcher.
P710 is a SBC, so there is no VIBRATOR, no battery, no sensor, no modem, maybe I should remove these service in device.mk ?

Sure, it's better to copy PPP or OPI4 directory and remove all unused stuff.

andyshrk, 2022-05-04:
Thanks,I can have a lot of fun with this great work now。

closed 2022-05-04 by andyshrk