GloDroid/glodroid_manifest #202: deploy-sd.img does not boot on pinephone pro

BurnyBoi, 2022-04-30:
I've copied deploy-sd.img onto 2 different micro SDs, and can't get it to start on the pinephone pro. I use TowBoot, and when attempting to boot from SD, towboot vibrates follows by flashing the red light a few times to indicate the boot from SD is not successful, and then boots from the emmc instead.

rsglobal, 2022-04-30:
Once deploy_sd sdcard inserted and phone booted you will see nothing on PPP (black screen, black LED).

But device should appear as USB download gadget and you should be able to continue flashing using `flash_sd.sh` script.

In future releases we plan to add some signs of life for bootloader state (at least green led).

McUberStein, 2022-05-02:
> Once deploy_sd sdcard inserted and phone booted you will see nothing on PPP (black screen, black LED).
>
> But device should appear as USB download gadget and you should be able to continue flashing [...]

I think you're misunderstanding. The phone does not boot from the SD card at all, it boots from the emmc. I have this same exact issue with TowBoot. I insert the SD card, I turn thr phone on, hold the volume down button to boot from SD, the LED glows blue for a second and then flashes red which means the phone cannot boot from the SD, and it boots from emmc. There is no black screen or anything, it just boots the OS installed on the emmc.

rsglobal, 2022-05-02:
Yes you're right. To boot it first time you have to erase bootloader from emmc.

McUberStein, 2022-05-03:
Got it. If you have TowBoot installed, you need to first get the TowBoot spi-flasher.img that you used to install it, flash is to an SD card, select something like "remove SPI contents", then power off, insert the Glodroid recovery SD, plug it into a PC and run the bash script. When the script is done you just have to wait for a minute or two (the phone screen will be black during that time) and then the Android logo will appear and it will boot. The problem is that Glodroid and TowBoot cannot coexist. so I do not think there is a way to boot back to the emmc without reinstalling TowBoot. Glodroid works perfectly though, and even though Azur Lane doesn't work right now and VLC seems to be stuck in the "Android TV" mode, the UI is extremely smooth and responsive. I haven't tried calling yet. but this is already a HUGE step for the PinePhone community! Keep up the good work!

rsglobal, 2022-05-03:
Sunxi has boot priority SD, EMMC.
Rockchip has boot priority EMMC, SD.

So by default in case EMMC has bootloader installed you have no chance to boot from SD. I believe TOWBOOT is just trying to boot linux system from SD partition when user is holding powerup or powerdown button, not the complete bootloader (Which GloDroid require to do).

But there is one trick that we've discovered while playing armbian on OPI4 board. It has boot priority SD->EMMC. Further investigation showed that proper TPL/SPL chain CAN detect bootloader presence on SDCARD and in such case start loading bootloader from SD-CARD.

GloDroid copied such SPL/TPL approach from ARMBIAN so in theory you can boot Linux from SDCARD even with GloDroid installed on EMMC (I haven't tried though)

So in case TowBoot guys will copy such SPL/TPL approach, GloDroid should be able to boot from SDCard even if TowBoot is installed on EMMC.

andyshrk, 2022-05-04:
We have a chance to change the boot priority on Rockchip at spl : that means modifying the `u-boot,spl-boot-order` on the emmc, let sdmmc in the front,  when you insert a bootable sdcard next time, it will try boot ATF,U-boot on sdmmc first    
```
andy@B150:~/WorkSpace/U-BOOT/u-boot$ rg "u-boot,spl-boot-order" arch/arm/dts/
arch/arm/dts/rk3328-roc-cc-u-boot.dtsi
10:             u-boot,spl-boot-order = "same-as-spl", &sdmmc, &emmc;

arch/arm/dts/rk3399-rock960-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3326-odroid-go2-u-boot.dtsi
8:              u-boot,spl-boot-order = &sdmmc;

arch/arm/dts/px30-u-boot.dtsi
13:             u-boot,spl-boot-order = &emmc, &sdmmc;

arch/arm/dts/rk3328-u-boot.dtsi
14:             u-boot,spl-boot-order = &emmc, &sdmmc;

arch/arm/dts/rk3368-lion-haikou-u-boot.dtsi
16:             u-boot,spl-boot-order = &emmc, &sdmmc;

arch/arm/dts/rk3399-evb-u-boot.dtsi
12:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3288-tinker-s-u-boot.dtsi
11:             u-boot,spl-boot-order = \

arch/arm/dts/rk3399-khadas-edge-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3399-rock-pi-4-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3399-firefly-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3288-u-boot.dtsi
11:             u-boot,spl-boot-order = \

arch/arm/dts/rk3308-evb-u-boot.dtsi
9:              u-boot,spl-boot-order = "same-as-spl", &emmc;

arch/arm/dts/rk3399-ficus-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3399-leez-p710-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3568-u-boot.dtsi
13:             u-boot,spl-boot-order = &sdhci, &sdmmc0;

arch/arm/dts/rk3399-puma-haikou-u-boot.dtsi
26:             u-boot,spl-boot-order = \

arch/arm/dts/rk3399-roc-pc-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &spi_flash, &sdhci, &sdmmc;

arch/arm/dts/rk3399pro-rock-pi-n10-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;

arch/arm/dts/rk3328-rock64-u-boot.dtsi
10:             u-boot,spl-boot-order = "same-as-spl", &sdmmc, &emmc;

arch/arm/dts/rk3399-rockpro64-u-boot.dtsi
10:             u-boot,spl-boot-order = "same-as-spl", &spi_flash, &sdmmc, &sdhci;

arch/arm/dts/rk3328-nanopi-r2s-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdmmc, &emmc;

arch/arm/dts/rk3368-px5-evb-u-boot.dtsi
10:             u-boot,spl-boot-order = &emmc;

arch/arm/dts/rk3399-pinebook-pro-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &spiflash, &sdmmc;

arch/arm/dts/rk3288-veyron.dtsi
19:             u-boot,spl-boot-order = &spi_flash;

arch/arm/dts/rk3308-roc-cc-u-boot.dtsi
9:              u-boot,spl-boot-order = "same-as-spl", &emmc;

arch/arm/dts/rk3568-evb-u-boot.dtsi
11:             u-boot,spl-boot-order = "same-as-spl", &sdmmc0, &sdhci;

arch/arm/dts/rk3399-nanopi4-u-boot.dtsi
10:             u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;
```

BurnyBoi, 2022-05-07:
> Yes you're right. To boot it first time you have to erase bootloader from emmc.

Is there a way to make deploy-sd.img seen as bootable by TowBoot like the other Pinephone Pro software images are doing?

rsglobal, 2022-05-12:
Sure, it's possible to do.

rsglobal, 2023-03-30:
Unfortunately, GloDroid and probably other native Rockchip images are not supported by the tow-boot. You may try booting raw sdcard image https://github.com/GloDroidCommunity/pine64-pinephonepro/releases

closed 2023-03-30 by rsglobal