GloDroid/glodroid_manifest #203: Slight occasional stutter when playing `Poland can into Space` on the Pinephone running GloDroid v0.7.5

volkertb, 2022-05-17:
I spent some time playing my favorite open-source Android game [Poland can into Space](https://f-droid.org/en/packages/pl.sanszo.pcis/) on my Pinephone (UBports Community Edition with 2GB RAM) running GloDroid v0.7.5.

The good news is that it runs smoothly most of the time, with proper graphics, sound and vibration support.

However, I did notice that the game occasionally stutters, ever so briefly. This affects gameplay, since it's quite a fast-paced game, and the phone very briefly stops responding and missing user input during those stutters, which is enough to lose the game, even if happens just for a fraction of a second.

I know, I know, a casual game like this doesn't seem like the most important use case, but games are a good way to stress-test graphics, animation, sound, input lag, etc. 

I'm not 100% sure, but I don't recall experiencing these stutters when I tried out the same game on v0.7.4. If it would be useful in helping to troubleshoot this, I guess I could downgrade the phone to v0.7.4 and try playing the game on that version again. Let me know if you'd like me to test that more thoroughly.

rsglobal, 2022-05-29:
I've noticed `logd` write to serial console blocks all interrupt for 30-50mS  if `logd`scheduled on CPU0.

Can you try disabling serial console and check if that fixes the issue?

`adb root && adb shell "echo 3 > /proc/sys/kernel/printk"`

rsglobal, 2023-03-21:
The latest releases should have a silent serial console after a successful boot. If this is still an issue, please open another issue [HERE](https://github.com/GloDroidCommunity/pine64-pinephone/issues).

closed 2023-03-21 by rsglobal