GloDroid/glodroid_manifest #209: 0.7.6 hangs on boot with pinephone

manuboll22, 2022-07-18:
Hello
tried both sd and mmc, the flashing appears to be successful, but the screen remains black and then the device shuts off of its own.
On rare occasions I've seen the Android logo pop up, but only for a short while.

The serial log shows this:

```
init: Command 'copy /system/etc/prop.default /dev/urandom' action=init (/system/etc/init/hw/init.rc:124) took 0ms and failed: Could not read input file '/system/etc/prop.default': open() failed: No such file or directory
init: Command 'write /dev/blkio/blkio.weight 1000' action=init (/system/etc/init/hw/init.rc:260) took 2ms and failed: Unable to write to file '/dev/blkio/blkio.weight': open() failed: Permission denied
init: Command 'write /dev/blkio/background/blkio.weight 200' action=init (/system/etc/init/hw/init.rc:261) took 2ms and failed: Unable to write to file '/dev/blkio/background/blkio.weight': open() failed: Permission denied
init: Command 'write /dev/blkio/blkio.group_idle 0' action=init (/system/etc/init/hw/init.rc:263) took 2ms and failed: Unable to write to file '/dev/blkio/blkio.group_idle': open() failed: Permission denied
init: Command 'write /dev/blkio/background/blkio.group_idle 0' action=init (/system/etc/init/hw/init.rc:264) took 2ms and failed: Unable to write to file '/dev/blkio/background/blkio.group_idle': open() failed: Permission denied
Registered swp emulation handler
init: starting service 'logd'...
init: Created socket '/dev/socket/logd', mode 666, user 1036, group 1036
init: Created socket '/dev/socket/logdr', mode 666, user 1036, group 1036
init: Created socket '/dev/socket/logdw', mode 222, user 1036, group 1036
init: starting service 'lmkd'...
init: Created socket '/dev/socket/lmkd', mode 660, user 1000, group 1000
init: starting service 'servicemanager'...
init: starting service 'hwservicemanager'...
```

rsglobal, 2022-07-18:
Hi, can you attach full log please?

manuboll22, 2022-07-18:
> Hi, can you attach full log please?

Sure, [here it is](https://github.com/GloDroid/glodroid_manifest/files/9130509/glodroid_076_pinephone_serial.txt).

manuboll22, 2022-07-18:
> It may also be related to battery under-voltage.

Hmm? How does that happen?

>  Does v0.7.5 work?

I just tested it and it does. I see the screen flicker, but perhaps that a story for another day.

tg-x, 2022-08-03:
not sure if it's the same issue, after flashing it booted fine with screen on.
but since the following day the screen remains black during&after booting (button sounds and vibrations are there)
the screen works in other distros and there's enough battery charge, with power connected

rsglobal, 2022-08-03:
@manuboll22 ,

Unfortunately I can't reproduce your case on my PP.

@tg-x ,

On my PP I am able to set brightness level to minimum, which will disable back-light. Can you check that's not the case from your end?

tg-x, 2022-08-03:
> On my PP I am able to set brightness level to minimum, which will disable back-light. Can you check that's not the case from your end?

indeed that was the problem, managed to fix by blindly swiping and turning back on..
but would be good to somehow limit the minimum to a still visible range.

herbsmn, 2022-11-04:
> > On my PP I am able to set brightness level to minimum, which will disable back-light. Can you check that's not the case from your end?
> 
> indeed that was the problem, managed to fix by blindly swiping and turning back on.. but would be good to somehow limit the minimum to a still visible level.

I accidentally set my brightness level to the lowest setting on my PinePhone running GloDroid and it too disabled the back-light.

If anyone else encounters this issue, the trick I used was shining a flashlight onto the screen. It allowed me to see enough to be able to increase the brightness and get the back light turned back on

closed 2023-03-21 by rsglobal