GloDroid/glodroid_manifest #210: Is it possible to install this on the PinePhone Pro via TOW Boot?

PucklaMotzer09, 2022-07-23:
I flashed the `deploy-sd-for-emmc.img` of `gd-v0.7.5-pinephonepro-userdebug.tar.gz` onto the device while it was in USB Mass Storage mode, but when turning it on it vibrated 10 times and then shutdown.
Is it even possible to flash GloDroid using TOW Boot or do I need to install UBoot again?

I also tried to flash `deploy-sd.img`, but it had the same result.

candymannetje, 2022-08-01:
Tow-boot is not supported.

fossdd, 2022-08-18:
Is it on the todo to get supported? Tow-boot is the modern soulution for the PinePhone Pro's SPI and is even delivered as default SPI on new PPPs. So on these devices GloDroid is unable to install.

rsglobal, 2022-08-18:
Unfortunately we have no free resources now to work on tow-boot integration. I also suspect that major changes have to be done from towboot side. 

But we will accept necessary code changes to support tow-boot in case somebody offers it.

hamishmb, 2022-11-09:
This is unfortunately becoming a problem, as newer PPPs come with Tow-boot from the factory, like mine did. I don't really understand well enough how the boot process works, or how Android works, to be of any assistance.

hamishmb, 2022-11-09:
Out of interest, why don't we just have an SD card image to deploy to SD, rather than having to use adb and fastboot?

rsglobal, 2022-11-20:
1. With fastboot user don't need to remove/insert sdcard every time (only 1-st flash requires this)
2. With fastboot EMMC flashing is possible
3. With fastboot user can flash single partition only

hamishmb, 2022-11-20:
Makes sense. I need to go read the spec for what all the different images do.

rapidreporter, 2022-11-30:
> This is unfortunately becoming a problem, as newer PPPs come with Tow-boot from the factory, like mine did. I don't really understand well enough how the boot process works, or how Android works, to be of any assistance.

I'm running into the same problem. I would love to try out Glodroid, but I am a bit scared of flashing back to U-Boot. I couldn't find a good tutorial to do this on the pinephone pro and don't want to push my luck by blind trying and ending up with a bricked device. I would love to see tow-boot support. Besides this could anyone provide me with a foolproof tutorial to flash back to U-Boot? Can I brick it by trying or am I overprotective?

Thanks :)

rsglobal, 2023-03-30:
Unfortunately GloDroid and probably other native Rockchip images are not supported by the tow-boot.

hamishmb, 2023-03-31:
Could we leave the issue open until there's some kind of resolution?

rsglobal, 2023-03-31:
Sure, but opening this issue in the tow-boot issue tracker would be more productive.

hamishmb, 2023-03-31:
Okay, I'll file it there as well and link back here.

herbsmn, 2023-04-30:
> Out of interest, why don't we just have an SD card image to deploy to SD, rather than having to use adb and fastboot?

There is now an SD card image for the PinePhone Pro: https://github.com/GloDroidCommunity/pine64-pinephonepro/releases/download/2023w13/gd-lineageos-pinephonepro-userdebug-2023w13.sdcard.img.zip

0chroma, 2023-04-30:
just tried the sd card image, it still doesn't seem to work with tow boot but it should hopefully make support easier

rsglobal, 2023-08-03:
rk2aw blob from Megous looks very promising. After installing on PPP's eeprom it reverses the boot probe order to SD, EMMC.
I haven't tried it yet, but maybe someone here could do it and report feedback.
https://xff.cz/kernels/rk2aw/rk2aw-rk3399-pinephone-pro/

nonamedll, 2023-10-03:
I recently installed rk2aw on my PPP and can confirm that it boots just fine.