GloDroid/glodroid_manifest #211: How to support a usb gps reciver(ublox 7020) on rpi4?

aswjh, 2022-07-31:
rpi4:/ # ls /dev/tty*                                                                                          
/dev/tty  /dev/ttyACM0  /dev/ttyS0  /dev/ttyprintk

rpi4:/ # cat /dev/ttyACM0
$GPTXT,01,01,02,u-blox ag - www.u-blox.com*50
$GPTXT,01,01,02,HW  UBX-G70xx   00070000 *77
$GPTXT,01,01,02,ROM CORE 1.00 (59842) Jun 27 2012 17:43:52*59
$GPTXT,01,01,02,PROTVER 14.00*1E
$GPTXT,01,01,02,ANTSUPERV=AC SD PDoS SR*20
$GPTXT,01,01,02,ANTSTATUS=OK*3B
$GPTXT,01,01,01,NMEA unknown msg*58
$GPTXT,01,01,02,LLC FFFFFFFF-FFFFFFFD-FFFFFFFF-FFFFFFFF-FFFFFFF9*53

jojo2357, 2022-09-27:
I do not know anything about this project, but I have used a GPS over Serial on Rapbian before. I wrote my own software to just listen in and handle that information, and I know that some use [`gpsd`](https://gpsd.gitlab.io/gpsd/index.html). If this isn't helpful I apologize, just trying to help

rsglobal, 2023-03-30:
Closing due to inactivity. Feel free to reopen it if your issue is still relevant.

closed 2023-03-30 by rsglobal