GloDroid/glodroid_manifest #213: Possibility of having a hardened/debloated edition

cristiioan, 2022-08-23:
While trying GloDroid, I saw that some of googles spyware is still there and I will run tommorow Universal Android Debloater on it, but is there any plan to get rid of it by default?

herbsmn, 2022-08-23:
I think the open issue about porting LOS and Replicant manifest to work with GloDroid is somewhat related to this issue. https://github.com/GloDroid/glodroid_manifest/issues/74

rsglobal, 2023-03-30:
> While trying GloDroid, I saw that some of googles spyware is still there and I will run tommorow Universal Android Debloater on it, but is there any plan to get rid of it by default?

In addition to AOSP, LineageOS support was added recently(https://github.com/orgs/GloDroidCommunity/repositories). At this moment PP and PPP are available only, but more devices will be added soon. Hope this will solve your issue.

closed 2023-03-30 by rsglobal

cristiioan, 2023-04-07:
> Can you describe it in detail? Which software you're referring to as spyware?
I will have to look again. I'm not sure and I haven't used it since last year.