GloDroid/glodroid_manifest #214: System always freezes when building with "make images"

IMSOASIAN, 2022-08-27:
_Hello there,_

I know that this is probably not the proper place to put this, but this is the only avenue I can utilize. The official AOSP mailing lists state that I cannot expect help for other distributions, and the discord link does not work for me.

### **OS: GNU/Linux Debian 11 "Bullseye" + certain testing 12 "Bookworm" packages**

### **RAM: 8GB DDR2**

### **Target Device: PinePhone**

### **Command entered: "make -j1 images"**

### **Debug log: Not shown, unless you want my mail**

### **Frozen prompt: "[   0% 0/1] analyzing Android.bp files and generating ninja file at out/soong/build.ninja"**

I had to build from source using an external hard drive since the download is upwards of 150GB. I do not believe this to be the cause of the issue.

The reason I had to use "-j1" is because building AOSP takes up a lot of RAM, so I limited it to one process (About [4GB of RAM](https://stackoverflow.com/questions/40925991/error-on-compiling-aosp-make-out-build-aosp-arm-ninja-killed#:~:text=one%20more%20thread%2C%204G%20more%20memory))

I cannot get a more meaningful output since the AOSP version of make is modified to not accept the "-d" or "--debug" options. I can't use regular make since it has not been modified to find the necessary build files for AOSP.

Please inform me if there is any more information I can give, I did all of the steps according to the guide (except for the "j1" part) and it's not working.

_Thanks,_

-[T] IMSOASIAN (For some reason the prebuilt for PinePhone still had serial console enabled?)

rsglobal, 2022-08-27:
Hi,

Building AOSP 12 and later requires 32GB of RAM.

System freezes because it starts swapping pages back and forth during ninja file preparation.
In theory, if you wait long enough, it should proceed further.

IMSOASIAN, 2022-08-27:
Got it. So should I increase my swap partition to combat this? I can put one on the same external hard drive.

rsglobal, 2022-08-28:
You'll need modern PC/laptop with 32gb of RAM to build AOSP.

closed 2022-08-28 by rsglobal