GloDroid/glodroid_manifest #215: Unable to flash glodroid to pinephone pro's sd card

Pethon, 2022-08-29:
I'm using pinephone pro developer model. I already flashed `deploy-sd.img` to sd card first. After I connected pinephone pro to macbook, I ran `./flash-sd.sh` which got an error below.
```
 $ ./flash-sd.sh
+ fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.008s]
Writing 'gpt'                                      OKAY [  0.029s]
Finished. Total time: 0.071s
+ fastboot flash bootloader bootloader-sd.img
Sending 'bootloader' (16352 KB)                    OKAY [  0.747s]
Writing 'bootloader'                               OKAY [  3.144s]
Finished. Total time: 3.914s
+ fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.013s]
Writing 'uboot-env'                                OKAY [  0.052s]
Finished. Total time: 0.093s
+ fastboot reboot
Rebooting                                          OKAY [  0.001s]
Finished. Total time: 0.001s
+ set +x
+ fastboot oem format
ERROR: could not clear input pipe; result e00002ed, ignoring...
ERROR: could not clear output pipe; result e00002d8, ignoring....
ERROR: usb_write failed with status e00002d9
FAILED (Write to device failed (No such file or directory))
fastboot: error: Command failed
```

Please help. Thank you.

rsglobal, 2022-08-29:
Hi, 

Can you share UART output?

Pethon, 2022-09-09:
@rsglobal 
Sorry for lacking of reply. But I have no idea what is UART and how to get it.
Can you guide me some command to get that output, or give me some link where I can read about this UART.

Thank you.

rsglobal, 2022-09-09:
https://pine64.com/product/pinebook-pinephone-pinetab-serial-console/

rsglobal, 2023-03-30:
The PPP project has moved to https://github.com/GloDroidCommunity/pine64-pinephonepro
Please check the latest release and reopen the issue if it is still relevant.

closed 2023-03-30 by rsglobal