GloDroid/glodroid_manifest #220: Pinephone beta edition - camera doesn't work

Saddamus, 2022-09-06:
There is no to much info on release notes. Is the camera issue related strictly to beta edition device or its general and known ? I'm just curious if there are some efforts to make it working.

rsglobal, 2022-09-07:
Hello @Saddamus ,

Camera isn't working for all devices except RPI4 where camera works but only partially.
At this moment I'm focused on graphics pipeline improving / hardening (mesa3d+drm_hwcomposer+gralloc components). Camera and codecs are in low priority.

herbsmn, 2022-11-21:
There is a lot going on in this waydroid issue about getting the cameras to work on the pinephone pro: https://github.com/waydroid/waydroid/issues/519