GloDroid/glodroid_manifest #221: Note: Support Matrix for PinePhone Pro

hamishmb, 2022-09-14:
As discussed on the Discord server, here's a feature Matrix of what is known to work on the PinePhone Pro. I don't (yet) have one, so I can't do any testing myself yet. Eventually, there'll probably be some wiki pages for this.

- Screen: Working
- GPU (graphics acceleration): Unknown
- Calls: Unknown
- SMS: Unknown
- 2G data networks: Unknown
- 3G/HSPA: Unknown
- 4G/LTE: Unknown
- Front-Facing Camera: Unknown
- Rear Camera: Unknown
- Other sensors: Unknown
- Waydroid: Working, no GPU acceleration

I'm sure there are omissions in this list. Please comment below with test results and the version of Glodroid you're running so I can add them to the list.

hamishmb, 2022-09-14:
Note, assuming cameras don't work due to https://github.com/GloDroid/glodroid_manifest/issues/220

rsglobal, 2022-09-14:
GPU GLES - works
GPU Vulkan - NO
Sensors - accel, gyro - works
Modem (ALL) - NO
WLAN should work
USBC should work in v0.8.1 (not released yet)

hamishmb, 2022-09-14:
Thanks, updated.

rsglobal, 2023-03-30:
Support matrix for the PPP moved here: https://github.com/GloDroidCommunity/pine64-pinephonepro/issues/1

closed 2023-03-30 by rsglobal