GloDroid/glodroid_manifest #222: Pinephone GloDroid fails to boot from EMMC

miklosakos, 2022-09-16:
Whenever I attempt to flash GloDroid on my Pinephone using the EMMC specific method, i get as far as getting the images flashed using the recovery SD, after removing the SD card and booting the phone back up it vibrates and the notification LED flashes red, phone does not appear as a usb device. Flashing to an SD Card works fine. I have this issue on 0.7.6, 0.7.8 and 0.8.0.

herbsmn, 2022-11-03:
I just flashed v.0.8.1 to my pinephone using the EMMC method and it worked just fine. This was my first attempt flashing GloDroid, so I can't say whether the versions you would work on my pinephone or not.

Mr-P-Butthole, 2022-12-09:
> I just flashed v.0.8.1 to my pinephone using the EMMC method and it worked just fine. This was my first attempt flashing GloDroid, so I can't say whether the versions you would work on my pinephone or not.

Can confirm for the first try. I suspect an issue with the formatting/reset from fastboot due to error about missing ext2fs

miklosakos, 2022-12-10:
I tried doing it again with v0.8.1 and the same issue persists, here's the process I did: dd "deploy-sd-for-emmc.img" onto sd card, power up the phone with the SD Card, run flash-emmc.sh, powering down the phone, removing recovery sd card and powering it up.
Console output doesn't look problematic.
```
miklos_akos@ghettomox:~/gd$ ./flash-emmc.sh 
+ ./fastboot flash gpt deploy-gpt.img
Sending 'gpt' (128 KB)                             OKAY [  0.013s]
Writing 'gpt'                                      OKAY [  0.030s]
Finished. Total time: 0.168s
+ ./fastboot flash bootloader bootloader-emmc.img
Sending 'bootloader' (1920 KB)                     OKAY [  0.106s]
Writing 'bootloader'                               OKAY [  0.109s]
Finished. Total time: 0.225s
+ ./fastboot flash uboot-env env.img
Sending 'uboot-env' (256 KB)                       OKAY [  0.019s]
Writing 'uboot-env'                                OKAY [  0.020s]
Finished. Total time: 0.051s
+ ./fastboot reboot
Rebooting                                          OKAY [  0.000s]
Finished. Total time: 0.654s
+ set +x
Power-down, remove recovery SD-CARD, and power-up the board.
Press enter to continue.
```

rsglobal, 2022-12-10:
Hm... Have you pressed enter to continue?

miklosakos, 2022-12-12:
> Hm... Have you pressed enter to continue?

There's no need, the phone cannot boot into Android Fastboot on the eMMC. The notification LED flashes red a few times and the phone vibrates as well before giving up and turning off, `lsusb` won't show the device either in fastboot mode.

rsglobal, 2022-12-12:
> The notification LED flashes red a few times and the phone vibrates as well before giving up and turning off,

That sound like your phone still has a different firmware.

There're 2 deploy images.
- deploy-sd.img
- deploy-sd-for-emms.img

Please check you're using the second one, otherwise it won't flash the EMMC.

miklosakos, 2022-12-12:
I tried both images, both result in the same error. Tried manually flashing [per this comment](https://github.com/GloDroid/glodroid_manifest/issues/32#issuecomment-655392444), however that didn't work either.

rsglobal, 2022-12-13:
We need more information. It would be nice to have serial console output of the flashing process.

miklosakos, 2022-12-16:
I'll have to order a serial UART cable then. Let's do keep this report open until then.
I can add that my PinePhone came with KDE Plasma on Manjaro which would correlate to revision/version 1.2 iirc.

miklosakos, 2023-03-14:
GloDroid 0.8.2 after running flash-emmc.sh, removing the sd card and booting the phone back up:
```
U-Boot SPL 2021.10 (Oct 04 2021 - 15:09:26 +0000)
DRAM: 2048 MiB
Trying to boot from MMC2
NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 00:00:00, Jan  1 1980
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x209cc00, model: Pine64 PinePhone (1.2)

Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)004 [variant: mmcboot]

CPU:   Allwinner A64 (SUN50I)
Model: Pine64 PinePhone (1.2)
DRAM:  2 GiB
MMC:   mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Please press [ESCAPE] or [CTRL+C] to enter the boot menu.
off
off
switch to partitions #0, OK
mmc1(part 0) is current device
MMC: no card present
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
No ethernet found.
No ethernet found.

ERROR: Could not boot anything from distro_bootcmd.

poweroff ...
```

rsglobal, 2023-03-15:
It looks like your EMMC still has an old bootloader on a second hardware partition. Please erase it and try again.
I tried looking for a guide for you by googling without any success.

In a few words, you must select emmc, switch to the second hardware partition and write it with random info or zeros using `mmc` uboot commands.

miklosakos, 2023-03-15:
Thank you for the help and sorry for being a n00b.
If anyone else runs into this: get [tow-boot](https://github.com/Tow-Boot/Tow-Boot/releases/download/release-2021.10-005/pine64-pinephoneA64-2021.10-005.tar.xz), burn it onto an sd card, boot it up and just hit erase. This will render the PinePhone unbootable without an SD card which is a non-issue since GloDroid ships it's own U-Boot.

closed 2023-03-15 by miklosakos