GloDroid/glodroid_manifest #228: Failure to flash 0.8.0 to PinePhone

KenDickey, 2022-10-21:
[flash.out.txt](https://github.com/GloDroid/glodroid_manifest/files/9840257/flash.out.txt)

I invoke `flash-sd.sh` from unpacked `gd-v0.8.0-pinephone-userdebug.tar.gz`
Running Raspian Linux on RasPi4. 
Did
  sudo apt-get install adb android-libext4-utils fastboot
to get aarch64/arm64 versions of fastboot (x86 version did not work well ;^)

After flash-sd.sh script 
 Notes from "flash.out.txt":
    Invalid sparse file format at header magic
    ...
    Warning: userdata type is raw, but ext4 was requested for formatting.
    /usr/lib/android-sdk/platform-tools/mke2fs failed with status 1
    fastboot: error: Cannot generate image for userdata

Symptom is that reboot shows green status light on PinePhone but screen remains blank and PinePhone unresponsive.

Any help?
Thanks a bunch!
-KenD

KenDickey, 2022-10-21:
Same symptom with `sudo`

[fastboot.bug.txt](https://github.com/GloDroid/glodroid_manifest/files/9841192/fastboot.bug.txt)

Other suggestions?

KenDickey, 2022-10-21:
Hmmm.. local dir's `mke2fs` is also an x86 binary.

Did a sim link

Raspian:RasPi4:~/Downloads/GoDroid >>> sudo ls -lasth /usr/lib/android-sdk/platform-tools/mke2fs
0 lrwxrwxrwx 1 root root 16 Oct 21 14:55 /usr/lib/android-sdk/platform-tools/mke2fs -> /usr/sbin/mke2fs

But error still exists.. 

Raspian:RasPi4:~/Downloads/GoDroid >>> sudo ./fastboot format:ext4 userdata
Warning: userdata type is raw, but ext4 was requested for formatting.
mke2fs 1.46.2 (28-Feb-2021)
/tmp/TemporaryFile-CPYZWd: Unimplemented ext2 library function while setting up superblock
/usr/lib/android-sdk/platform-tools/mke2fs failed with status 1
fastboot: error: Cannot generate image for userdata

rsglobal, 2022-10-22:
> Unimplemented ext2 library

Can you try to erase partitions instead of formatting? It worked for older version of Android, where system rebooted into recovery and does the formatting there automatically.

I.e. fastboot erase userdata

KenDickey, 2022-10-22:
Raspian:RasPi4:~/Downloads/GoDroid >>> sudo fastboot erase userdata
Erasing 'userdata'                                 OKAY [ 85.947s]
Finished. Total time: 85.988s

Raspian:RasPi4:~/Downloads/GoDroid >>> ./fastboot format:ext4 userdata
FAILED (remote: 'failed to set partition')
fastboot: error: Command failed
Raspian:RasPi4:~/Downloads/GoDroid >>> sudo ./fastboot format:ext4 userdata
FAILED (remote: 'failed to set partition')
fastboot: error: Command failed

Raspian:RasPi4:~/Downloads/GoDroid >>>

KenDickey, 2022-10-22:
Note that FastBoot screen does display on phone.
==========================
Android Recovery
Pine64/pinephone/pinephone
13/TPIA.220624.021.AI/eng.user.20220819.041310
user debug/test-keys
Use volume up-down and power
====
Can't load Android system. Your data may be corrupt ... ...
====
 Try Again
 Factory Data Reset
===========================

Tried the factory data reset as well.   Still no joy.

KenDickey, 2022-10-24:
Tried again with 0.8.1.  Still no joy.
[failure.out.txt](https://github.com/GloDroid/glodroid_manifest/files/9855376/failure.out.txt)

rsglobal, 2022-10-25:
Using RapsberryPI as a host isn't supported. Please try x86-based linux.

KenDickey, 2022-10-25:
On 2022-10-25 03:18, rsglobal wrote:

> Using RapsberryPI as a host isn't supported. Please try x86-based 
> linux.

Wow.  It has been quite a while.  I did find an old Intel x86 box in the 
closet.  Gigantic heat sinks and multiple fans. Gack!

Also, Ubuntu distro is 3.8 GB.  I'll try LUbuntu which at 2.7 GB is 
still bloatware, IMHO.

Sorry, I am used to Smalltalk, where I can run the same image -- pixel 
per pixel -- on Intel/ARM/RISCV under whatever OS.  I even built the VM 
on PinePhone using Manjaro and on a RISCV using Debian.

Sorry to hear that one still needs to use an x86 CPU to build/install 
for ARM.  I hope "self hosting" gets into your goal set at some point.

Looks like still an hour download to get the LUbuntu iso.  I'll let you 
know how this works.

Thanks much for the help.  Sorry for the rant.  It has been quite some 
time since I have used Intel CPUs.  My house is net-zero and I prefer to 
send the excess electrons to shut down peaker generating plants.  The 
ol' Intel/AMD box does warm the house, however.  ;^)

KenDickey, 2022-10-27:
OK. x86/64 was overheating, so after vacuuming out fan, replacing thermal paste under heat sinks, and installing a Linux (antiX), redownloading GoDroid et al..

I still get the symptom above.
============
+ ./fastboot erase misc
Erasing 'misc'                                     OKAY [  0.016s]
Finished. Total time: 0.038s
+ ./fastboot reboot-fastboot
Rebooting into fastboot                            OKAY [  0.006s]
< waiting for any device >
Finished. Total time: 569.307s
+ ./fastboot flash super super.img
Invalid sparse file format at header magic
Sending sparse 'super' 1/7 (262140 KB)             OKAY [ 25.154s]
Writing 'super'                                    FAILED (remote: 'No such file or directory')
fastboot: error: Command failed
============

herbsmn, 2022-11-03:
I didn't save my logs but I had a similar issue using flash-sd.sh on x86 when trying to install GloDroid v0.8.1 onto my Pinephone.

flash-emmc.sh ended up working fine, though, and since I wanted GloDroid on my emmc anyways I didn't investigate the issue with flash-sd.sh failing while writing 'super' any further.

KenDickey, 2022-11-04:
Thanks for this!  
I was able to dd the deploy-sd-for-emmc.img onto the SD and to successfully run the flash-emmc.sh script.
A looooo..oong time after the green light, I noted the Android boot screen.
Now I have to figure out WTH is Android!  :^)

KenDickey, 2022-11-04:
Works with Intel Linux connected & emmc script.

closed 2022-11-04 by KenDickey

herbsmn, 2022-11-05:
Happy to help! Now makes me wonder if the flash-emmc.sh script works on arm too.

herbsmn, 2022-11-05:
Also, it seems like we have identified some sort of bug with flash-sd.sh for the PinePhone on 0.8.0 and 0.8.1. Maybe we should change the name of the issue to reflect that and open it back up.

KenDickey, 2022-11-07:
On 2022-11-05 09:33, FreeKurt.com wrote:

> Happy to help! Now makes me wonder if the flash-emmc.sh script works on 
> arm too.

Nope.  Tried that first.  "./fastboot format:ext4 userdata" still fails 
on arm64 linux.