GloDroid/glodroid_manifest #230: [RPI4] Cannot boot to Home Scer

nguyenanhgiau, 2022-12-28:
Hi guys,

Branch master
Android OS: 13
Device: Raspberry Pi 4 - 4G RAM
Below is the console log

[   74.718557] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:                                                                                                                                       
[   74.724753]  (detected by 2, t=15290 jiffies, g=601, q=119)                  
[   74.730402] rcu: All QSes seen, last rcu_preempt kthread activity 15282 (4294910976-4294895694), jiffies_till_next_fqs=1, root ->qsmask 0x0
[   74.743102] rcu: rcu_preempt kthread timer wakeup didn't happen for 15281 jiffies! g601 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x200
[   74.754567] rcu:     Possible timer handling issue on cpu=3 timer-softirq=225                         
[   74.761534] rcu: rcu_preempt kthread starved for 15282 jiffies! g601 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x200 ->cpu=3
[   74.772029] rcu:     Unless rcu_preempt kthread gets sufficient CPU time, OOM is now expected behavior.
[   74.781286] rcu: RCU grace-period kthread stack dump:                                      
[   74.786403] task:rcu_preempt     state:R stack:    0 pid:   15 ppid:     2 flags:0x00000008
[   74.794877] Call trace:             
[   74.797349]  __switch_to+0x164/0x210
[   74.800980]  __schedule+0x564/0x850
[   74.804516]  schedule+0xa0/0xe4         
[   74.807696]  schedule_timeout+0xb4/0x1ac
[   74.811671]  rcu_gp_fqs_loop+0x168/0x740
[   74.815648]  rcu_gp_kthread+0x74/0x218
[   74.819444]  kthread+0x154/0x34c
[   74.822711]  ret_from_fork+0x10/0x20
[   74.826332] rcu: Stack dump where RCU GP kthread last ran:
[   74.831888] Task dump for CPU 3:
[   74.835153] task:kworker/u8:0    state:R  running task     stack:    0 pid:    8 ppid:     2 flags:0x00000008
[   74.845211] Workqueue: events_unbound deferred_probe_work_func
[   74.851129] Call trace:
[   74.853599]  __switch_to+0x164/0x210
[   74.857220]  0xffffff8042ef4080

rsglobal, 2022-12-28:
Hey,

Are you booting images from release page, or have you built it by yourself?

nguyenanhgiau, 2022-12-29:
I tried both and got the same problem.
release version: v0.8.1

rsglobal, 2022-12-29:
Could you attach full log please?

nguyenanhgiau, 2022-12-29:
Hi @rsglobal, this is the console log.
[rpi4_v801.log](https://github.com/GloDroid/glodroid_manifest/files/10319768/rpi4_v801.log)

rsglobal, 2022-12-29:
Thanks. 
Similar could happen when power is not enough.
Please try again using different cable / PS.

nguyenanhgiau, 2022-12-29:
omg, it works.
But from now, I cannot use adb command through USB connection.
Do you have any solution? adb through wifi?

rsglobal, 2022-12-29:
1. Try using quality cable, i.e. hoco 5A. Ensure you're not powering anything from RPI4 board (no embedded displays, hard disks, etc)
2. Wifi should also work, see https://docwiki.embarcadero.com/RADStudio/Alexandria/en/Android_Device_Wireless_Debugging , Steps to pair Android 11 and above section.

closed 2022-12-29 by nguyenanhgiau