GloDroid/glodroid_manifest #234: Unable to flash pinephone with latest glodroid

h-n-a, 2023-02-26:
Tried to install the latest glodroid release but it fails right at the bootloader menu and when i try to enter to fastboot mode

[image]
[image]

herbsmn, 2023-03-11:
I had a similar issue with the last release. I was able to get it to work by using the flash emmc method instead of the flash sd method. Any chance you'd be willing to try that out and see if it works for you?

h-n-a, 2023-03-11:
@herbsmn thanks for reaching out, yeah i'll try it later today or tomorrow and let you know.

rsglobal, 2023-03-21:
Please check this PR: https://github.com/GloDroidCommunity/pine64-pinephone/pull/11
(there's a test image link in the first comment)

If the issue is still relevant, please open another issue [HERE](https://github.com/GloDroidCommunity/pine64-pinephone/)

closed 2023-03-21 by rsglobal