GloDroid/glodroid_manifest #243: RPI4 usb storage boot fail

gerbit, 2024-06-15:
I'm trying to bring up full boot from SSD drive connected over USB. So i've prepared u-boot:
`CONFIG_BOOTCOMMAND="setenv intf usb && setenv bootdev 0 && setenv scriptaddr 0x3000000 && load ${intf} ${bootdev}:2 ${scriptaddr} boot.scr && source ${scriptaddr}"`

<details>
  <summary>generated boot.txt</summary>

```shell
setenv dtbaddr 0x1fa00000;
setenv dtb_index 0x0
setenv partitions "uuid_disk=\${uuid_gpt_disk}"
setenv partitions "${partitions}"";name=bootloader,start=128K,size=130944K,uuid=\${uuid_gpt_bootloader}"
setenv partitions "${partitions}"";name=uboot-env,size=512K,uuid=\${uuid_gpt_reserved}"
setenv partitions "${partitions}"";name=misc,size=512K,uuid=\${uuid_gpt_misc}"
setenv partitions "${partitions}"";name=frp,size=512K,uuid=\${uuid_gpt_frp}"
setenv partitions "${partitions}"";name=boot_a,size=64M,uuid=\${uuid_gpt_boot_a}"
setenv partitions "${partitions}"";name=boot_b,size=64M,uuid=\${uuid_gpt_boot_b}"
setenv partitions "${partitions}"";name=init_boot_a,size=8M,uuid=\${uuid_gpt_init_boot_a}"
setenv partitions "${partitions}"";name=init_boot_b,size=8M,uuid=\${uuid_gpt_init_boot_b}"
setenv partitions "${partitions}"";name=vendor_boot_a,size=32M,uuid=\${uuid_gpt_vendor_boot_a}"
setenv partitions "${partitions}"";name=vendor_boot_b,size=32M,uuid=\${uuid_gpt_vendor_boot_b}"
setenv partitions "${partitions}"";name=vbmeta_a,size=512K,uuid=\${uuid_gpt_vbmeta_a}"
setenv partitions "${partitions}"";name=vbmeta_b,size=512K,uuid=\${uuid_gpt_vbmeta_b}"
setenv partitions "${partitions}"";name=vbmeta_system_a,size=512K,uuid=\${uuid_gpt_vbmeta_system_a}"
setenv partitions "${partitions}"";name=vbmeta_system_b,size=512K,uuid=\${uuid_gpt_vbmeta_system_b}"
setenv partitions "${partitions}"";name=super,size=""3000""M,uuid=\${uuid_gpt_super}"
setenv partitions "${partitions}"";name=metadata,size=16M,uuid=\${uuid_gpt_metadata}"
setenv partitions "${partitions}"";name=userdata,size=-,uuid=\${uuid_gpt_userdata}"
setenv bootargs " init=/init rootwait ro androidboot.boottime=223.708 androidboot.selinux=permissive"
setenv bootargs "${bootargs}"" androidboot.revision=1.0 androidboot.board_id=0x1234567 androidboot.serialno=${serial#}"
setenv bootargs "${bootargs}"" firmware_class.path=/vendor/etc/firmware"
setenv bootargs "${bootargs}"" ${debug_bootargs} printk.devkmsg=on quiet androidboot.workaround=is_here"
setenv enter_fastboot '
 setenv fastboot_fail 0
 fastboot 0 || setenv fastboot_fail 1;
 if test \"${fastboot_fail}\" = \"1\";
 then
  part start \{$intf} \{$bootdev} misc misc_start || gpt write $partitions
  bcb load \{$intf} \{$bootdev} misc && bcb set command boot-fastboot && bcb store
 fi;
'
setenv bootcmd_bcb '
 ab_select slot_name \${intf} \${bootdev}#misc --no-dec || run enter_fastboot ;
 bcb load \${intf} \$bootdev misc ;
 bcb test command = bootonce-bootloader && bcb clear command && bcb store && run enter_fastboot ;
 bcb test command = boot-fastboot && setenv androidrecovery true ;
 bcb test command = boot-recovery && setenv androidrecovery true ;
 if test \"\${androidrecovery}\" != \"true\";
 then
  ab_select slot_name \${intf} \${bootdev}#misc || run enter_fastboot ;
 fi;
 setenv bootargs "\${bootargs}"" androidboot.slot_suffix=_\$slot_name" ;
'
setenv avb_verify '
 avb init \$bootdev; avb verify _\$slot_name;
'
setenv bootcmd_avb '
 setenv bootargs "${bootargs}"" androidboot.verifiedbootstate=orange "
'
setenv bootcmd_prepare_env '
 if test \"\${intf}\" = \"usb\"; then
  usb dev 0;
  setenv bootdevice_path \"scb/fd500000.pcie\";
 else
  setenv bootdevice_path \"emmc2bus/fe340000.mmc\";
 fi
 setenv bootargs "\${bootargs}"" androidboot.boot_devices=\${bootdevice_path}" ;
'
setenv bootcmd_start '
 if test \"\${androidrecovery}\" != \"true\";
 then
  setenv bootargs "\${bootargs}"" androidboot.force_normal_boot=1" ;
 fi;
 abootimg addr \$abootimg_boot_ptr \$abootimg_vendor_boot_ptr \$abootimg_init_boot_ptr
 fdt addr \${fdt_addr} &&
 bootm \$abootimg_boot_ptr
'
setenv bootcmd_block '
 run bootcmd_bcb
 abootimg load \${intf} \$bootdev boot \${slot_name}
 abootimg load \${intf} \$bootdev init_boot \${slot_name}
 abootimg load \${intf} \$bootdev vendor_boot \${slot_name}
 if test \"\$androidrecovery\" = \""true"\";
 then
  setenv bootargs "${bootargs}"" androidboot.verifiedbootstate=orange ";
 else
  run bootcmd_avb;
 fi;
'
setenv rename_and_expand_userdata_placeholder '
  part number \${intf} \${bootdev} userdata_placeholder partition_number
  if test -n "${partition_number}";
  then
    echo "Renaming userdata_placeholder partition to userdata...";
    gpt read \${intf} \${bootdev} current_layout
    setexpr new_layout gsub "name=userdata_placeholder" "name=userdata" ${current_layout}
    gpt write \${intf} \${bootdev} ${new_layout}
    echo "The userdata_placeholder partition has been renamed to userdata.";
    echo "Expanding userdata partition to fill the entire drive...";
    gpt read \${intf} \${bootdev} expanded_layout
    setexpr final_layout gsub "name=userdata,start=[^,]*,size=[^,]*,uuid" "name=userdata,start=[^,]*,size=-,uuid" ${expanded_layout}
    gpt write \${intf} \${bootdev} ${final_layout}
    echo "The userdata partition has been expanded.";
  fi;
'
setenv bootcmd '
 run bootcmd_prepare_env ;
 run rename_and_expand_userdata_placeholder ;
 run bootcmd_block ;
 run bootcmd_start ;
'
run bootcmd
reset
``` 

</details>

And the output of this: kernel failed to load:

<details>
  <summary>crash</summary>

```shell
## Executing script at 03000000

Device 0: Vendor: JMicron  Rev: 0214 Prod: Tech
            Type: Hard Disk
            Capacity: 238475.1 MB = 232.8 GB (488397168 x 512)
... is now current device
rename_and_expand_userdata_placeholder...
Renaming userdata_placeholder partition to userdata...
success!
Writing GPT: success!
The userdata_placeholder partition has been renamed to userdata.
Expanding userdata partition to fill the entire drive...
success!
Writing GPT: success!
The userdata partition has been expanded.
bootcmd_block...
bootcmd_bcb...
ANDROID: Invalid CRC-32 (expected 807077e9, found 00000000),re-initializing A/B metadata.
ANDROID: Attempting slot a, tries remaining 7
ANDROID: Booting slot: a
ANDROID: Attempting slot a, tries remaining 7
ANDROID: Booting slot: a
Loaded Android boot image using smart load (18/64 MB)
Loaded 'boot' partition to address 0x00000000320f7000 (size: 0x4000000) in 107 ms
Loaded Android boot image using smart load (3/8 MB)
Loaded 'init_boot' partition to address 0x00000000360f8000 (size: 0x800000) in 21 ms
Loaded Android boot image using smart load (16/32 MB)
Loaded 'vendor_boot' partition to address 0x00000000368fa000 (size: 0x2000000) in 74 ms
bootcmd_start...
Working FDT set to 2eff1f00
## Booting Android Image at 0x320f7000 ...
Kernel load addr 0x00080000 size 17557 KiB
Kernel extra command line: androidboot.hardware=sxm_rpi4_car coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 console=ttyS0,115200
RAM disk load addr 0x03300000 size 17674 KiB
For boot header v3+ vendor boot image has to be provided
For boot header v3+ vendor boot image has to be provided
Working FDT set to 2eff1f00
"Synchronous Abort" handler, esr 0x96000004, far 0xcfcfcdcb00dd2730
elr: 00000000000a8a4c lr : 00000000000b3194 (reloc)
elr: 000000003b059a4c lr : 000000003b064194
x0 : 3030323531312c30 x1 : 3030323531312c30
x2 : 000000003b0d2000 x3 : 0000000000000000
x4 : cfcfcdcb00dd2720 x5 : 00000000320e5350
x6 : 0000000000000000 x7 : 000000003b0d2878
x8 : 00000000388fa000 x9 : 0000000000000004
x10: 0000000032038288 x11: 0000000000000000
x12: 0000000000000200 x13: 000000003e8d8560
x14: 00000000ffffffff x15: 000000003201dbf8
x16: 0000000000000000 x17: 0000000000000000
x18: 000000003202cd60 x19: 0000000000000000
x20: 00000000320e18f0 x21: 0000000000000000
x22: 000000003100b040 x23: 0000000000000001
x24: 00000000ffffff7f x25: 000000003b0e9460
x26: 00000000320f7000 x27: 00000000368fa000
x28: 00000000368fa000 x29: 000000003201d950

Code: f85f0000 910040e7 cb000084 8b000021 (f9400880)
Resetting CPU ...
```

</details>

Changing 'intf' from 'usb' to 'mmc' makes sdcard boot fine. And I wonder, what's wrong with usb storage. Any hints?

gerbit, 2024-06-18:
Update:

decision made to leave boot, vendor_boot, init_boot, frp and misc on SD Card, all other partitions - on SSD. Works smooth and clean.

rsglobal, 2024-06-18:
AFAIK the Tesla-Android project did that, and it is based on GloDroid. So you may refer to it for inspiration.

gerbit, 2024-06-18:
@rsglobal yes, looks like tesla-android managed to get it work on cm4 with nvme drive, but not with rpi4. There is no way to attach nvme drive directly to rpi4 board, only over usb, but looks like u-boot XHCI driver corrupts memory when bootscript loads the boot image using `part read usb`. I've verified that the boot data lays in the memory correctly using md.w after `part  read usb` and your custom smart load implementation `abootimg load`. The same script but with `mmc read` works perfectly.
This is just a note. I don't know whom to address to this issue. Just leaving it here.